sukesh-ak/ESP32-TUX #2: Unknown CMake command "spiffs_create_partition_image"

paulodoux, 2022-11-07:
I'm unable to build with "idf.py -B build-esp32s3 set-target esp32s3 build" I'm getting the following error : 
```
CMake Error at main/CMakeLists.txt:17 (spiffs_create_partition_image):
  Unknown CMake command "spiffs_create_partition_image".

-- Configuring incomplete, errors occurred!
```

sukesh-ak, 2022-11-07:
- IDF version?
- OS version?
- Where are you compiling? from IDF command prompt?

I hope you are using the latest source. 
Are you compiling from IDF command line? Which OS?

SPIFF is created from the folder called 'fatfs'.
To get that compiled and uploaded, the cmake line is [this](https://github.com/sukesh-ak/ESP32-TUX/blob/f71e0b5b4d582b09202939d8d71ddab5d6dc8904/main/CMakeLists.txt#L17)

If you don't use both esp32 & esp32-s3 then you can just use the following. 
set-target is required only once. 

```cmd
idf.py set-target esp32s3
idf.py build
```

Try from the ESP-IDF command-line.

sukesh-ak, 2022-11-07:
Otherwise change the CMakeLists.txt from [here](https://github.com/sukesh-ak/ESP32-TUX/blob/f71e0b5b4d582b09202939d8d71ddab5d6dc8904/main/CMakeLists.txt#L17)

From 
```
spiffs_create_partition_image(storage ../fatfs FLASH_IN_PROJECT)
```
To
```
spiffs_create_partition_image(storage ${PROJECT_DIR}/fatfs FLASH_IN_PROJECT)
 
```

@paulodoux Please pull the latest commit (above fix) and see if it resolves the issue.

sukesh-ak, 2022-11-07:
There are few more things to take care of.
I am still waiting for my Plus version to test.

Change these in menuconfig
1. Change the Flash size to 4/16 as per your device spec
2. Partition Table select `Custom partition table CSV`
3. Enable PSRAM in `Component Config > ESP32-S3-specific > Support for external, SPI-connected RAM`

sukesh-ak, 2022-11-07:
Fixed in commit 8f0f54b23b0613613bdeefc7cdf8c8bf538b1ac9

closed 2022-11-07 by sukesh-ak

paulodoux, 2022-11-08:
Thanks, but it doesn't change anything .
• I'm running the last idf, on fresh install, test it on my Mac and a ubuntu VM.
• I'm compiling from the cmd line
• I'm compiling IDF example "storage/spiffs" with success (spiffs_create_partition_image)
• i've pulled the last commit
• Can't get to menuconfig

idf.py set-target esp32s3 give me : 

```
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/Users/paulodoux/esp/ESP32-TUX/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /Users/paulodoux/esp/ESP32-TUX/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/paulodoux/.espressif/python_env/idf5.1_py3.8_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /Users/paulodoux/esp/ESP32-TUX"...
-- Found Git: /usr/bin/git (found version "2.30.1 (Apple Git-130)") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/paulodoux/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/paulodoux/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/paulodoux/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Solving dependencies requirements
Updating lock file at /Users/paulodoux/esp/ESP32-TUX/dependencies.lock
Processing 2 dependencies:
[1/2] espressif/fmt (9.1.0)
[2/2] idf (5.1.0)
-- Project sdkconfig file /Users/paulodoux/esp/ESP32-TUX/build/sdkconfig
Loading defaults file /Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults...
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:189 CONFIG_MCPWM_ISR_IN_IRAM was replaced with CONFIG_MCPWM_ISR_IRAM_SAFE
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:232 CONFIG_ESP32S3_DEFAULT_CPU_FREQ_80 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_80
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:233 CONFIG_ESP32S3_DEFAULT_CPU_FREQ_160 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_160
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:234 CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:235 CONFIG_ESP32S3_DEFAULT_CPU_FREQ_MHZ was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:264 CONFIG_ESP32S3_SPIRAM_SUPPORT was replaced with CONFIG_SPIRAM
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:269 CONFIG_ESP32S3_DEBUG_OCDAWARE was replaced with CONFIG_ESP_DEBUG_OCDAWARE
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:270 CONFIG_ESP32S3_BROWNOUT_DET was replaced with CONFIG_ESP_BROWNOUT_DET
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:271 CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_7 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_7
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:272 CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_6 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_6
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:273 CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_5 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_5
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:274 CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_4 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_4
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:275 CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_3 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_3
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:276 CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_2 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_2
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:277 CONFIG_ESP32S3_BROWNOUT_DET_LVL_SEL_1 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_1
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:278 CONFIG_ESP32S3_BROWNOUT_DET_LVL was replaced with CONFIG_ESP_BROWNOUT_DET_LVL
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:279 CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_FRC1 was replaced with CONFIG_NEWLIB_TIME_SYSCALL_USE_RTC_HRT
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:280 CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC was replaced with CONFIG_NEWLIB_TIME_SYSCALL_USE_RTC
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:281 CONFIG_ESP32S3_TIME_SYSCALL_USE_FRC1 was replaced with CONFIG_NEWLIB_TIME_SYSCALL_USE_HRT
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:282 CONFIG_ESP32S3_TIME_SYSCALL_USE_NONE was replaced with CONFIG_NEWLIB_TIME_SYSCALL_USE_NONE
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:283 CONFIG_ESP32S3_RTC_CLK_SRC_INT_RC was replaced with CONFIG_RTC_CLK_SRC_INT_RC
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:284 CONFIG_ESP32S3_RTC_CLK_SRC_EXT_CRYS was replaced with CONFIG_RTC_CLK_SRC_EXT_CRYS
/Users/paulodoux/esp/ESP32-TUX/sdkconfig.defaults:285 CONFIG_ESP32S3_RTC_CLK_SRC_EXT_OSC was replaced with CONFIG_RTC_CLK_SRC_EXT_OSC
[... cut ...]
-- Required features: cxx_variadic_templates
CMake Error at main/CMakeLists.txt:17 (spiffs_create_partition_image):
  Unknown CMake command "spiffs_create_partition_image".

-- Configuring incomplete, errors occurred!
See also "/Users/paulodoux/esp/ESP32-TUX/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /Users/paulodoux/esp/ESP32-TUX/build/log/idf_py_stderr_output_3355 and /Users/paulodoux/esp/ESP32-TUX/build/log/idf_py_stdout_output_3355

```

sukesh-ak, 2022-11-08:
Project is setup for IDF version 4.x
IDF 5.x is not released and there are breaking changes in the framework as well.  Once its released I will make it compatible.

Till it's not released, I will stay with the released branch. I have other projects which are dependent on stable IDF release.

paulodoux, 2022-11-08:
thanks, compiling ok with 4.4.

sukesh-ak, 2022-11-08:
@paulodoux Thanks for confirmation. 
If you have any questions, you can use the discussion tab.

For bugs/issues feel free to open one here :)

sukesh-ak, 2022-11-08:
@paulodoux Also I believe there is some issue with spiffs_create_partition_image in IDF 5.
Not sure if they decided to change or add since there is a new option called `fatfs_create_spiflash_image`

paulodoux, 2022-11-08:
By the way I have to make small modification for compiling : 
- helper_storage.hpp:43 cast error
 esp_err_t ret = spi_bus_initialize(**SPI_HOST_ID**, &bus_cfg, SDSPI_DEFAULT_DMA);
- helper_storage.hpp:38 typo error
.sclk_io_num = SD_**S**CLK,

sukesh-ak, 2022-11-08:
@paulodoux Thanks. Actually I am still waiting for my device (Plus). 
So the code for it is not tested. Thanks for testing :)

Btw, I have also mapped the SPIFF to LVGL in lv_conf.h, so you can use F:/\<filename>

```c
/*API for open, read, etc*/
#define LV_USE_FS_POSIX 1
#if LV_USE_FS_POSIX
    #define LV_FS_POSIX_LETTER 'F'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_POSIX_PATH "/spiffs"         /*Set the working directory. File/directory paths will be appended to it.*/
    #define LV_FS_POSIX_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif

```

sukesh-ak, 2022-11-08:
@paulodoux 
Also if you have the 16MB Flash version, you can change the partition size in `partitions.csv`

paulodoux, 2022-11-09:
For your information, the SC01 plus is incorrectly sell as 16MB Flash : 
spi_flash: Detected size(8192k)

paulodoux, 2022-11-09:
[image]

sukesh-ak, 2022-11-09:
They have multiple variants. 16MB is available only from factory. That’s the link I have provided after confirming with them.