sukesh-ak/ESP32-TUX #6: Setup for Makerfabs 16Bit Parallel TFT 3.5" ILI9488 - ESP32-S3

hayschan, 2022-12-05:
## Problem

1. Lots of error showed in serial monitor.
2. After printing out the errors, the device will reset itself.
3. The display does not turn on.

## Situation

I am using the [ESP32-S3 Parallel TFT with Touch 3.5'' ILI9488](https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-ili9488.html).

First, I modified the device selection in main.hpp to include only my device, and I did not modify anything else.
```c++
/********************DEVICE SELECTION ******************/
/* Enable one of the devices from below (shift to bsp selection later) */
// #include "conf_WT32SCO1.h"              // WT32-SC01 (ESP32)

// WT32-SC01 Plus (ESP32-S3 + 8Bit Parellel) with SD Card, Audio support
// #include "conf_WT32SCO1-Plus.h"         

// Makerfabs ESP32S335D (ESP32-S3 + 16Bit Parellel) with SD Card, Audio support
#include "conf_Makerfabs_S3_PTFT.h"     

// Makerfabs ESP32S335D (ESP32-S3 + SPI) with SD Card, Audio support
// #include "conf_Makerfabs_S3_STFT.h"
/********************************************************/
```

Then, I followed [Flash](https://github.com/sukesh-ak/ESP32-TUX#separate-build-folder-for-esp32--esp32-s3) section of README.md

#### Flash log
```
Serial port /dev/cu.usbserial-026DC87A
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
...
Compressed 20832 bytes to 13280...
Writing at 0x00000000... (100 %)
Wrote 20832 bytes (13280 compressed) at 0x00000000 in 0.6 seconds (effective 261.8 kbit/s)...
Hash of data verified.
Compressed 2002688 bytes to 1127053...
Writing at 0x00010000... (1 %)
... 
...
...
Writing at 0x0064e59a... (100 %)
Wrote 524288 bytes (169669 compressed) at 0x00610000 in 5.7 seconds (effective 739.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
```

After flashing, I could see LOGs from the board through serial monitor, which indicates the flash is successful.

However, there are the following three problems:
1. Lots of error showed in serial monitor.
2. After printing out the errors, the device will reset itself.
3. The display does not turn on.

#### Error log
[... 113 lines cut ...]

0x421214a7: lgfx::v1::Panel_Device::init(bool) at /Users/hays/Github/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/panel/Panel_Device.cpp:60

0x42025a3b: lgfx::v1::Panel_LCD::init(bool) at /Users/hays/Github/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/panel/Panel_LCD.cpp:31

0x4202424e: lgfx::v1::LGFX_Device::init_impl(bool, bool) at /Users/hays/Github/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/LGFXBase.cpp:3099

0x4200ef7f: lgfx::v1::LGFX_Device::init() at /Users/hays/Github/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/LGFXBase.hpp:1138
 (inlined by) app_main at /Users/hays/Github/ESP32-TUX/main/main.cpp:267

0x4212a90b: main_task at /Users/hays/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

0x40381c71: vPortTaskWrapper at /Users/hays/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ELF file SHA256: 959501bcf6829e98

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375ad1
0x40375ad1: esp_restart_noos_dig at /Users/hays/esp/esp-idf/components/esp_system/esp_system.c:46 (discriminator 1)
```

At the last 5 lines, the device starts to reboot and the Error-Reboot loop continues on and on.

sukesh-ak, 2022-12-05:
@hayschan I just upgraded the project to IDF 5.0 which also needs `develop` branch of LovyanGFX.
LovyanGFX for Parallel TFT is broken for now and you can follow the issue [here](https://github.com/lovyan03/LovyanGFX/issues/322)

SPI devices are working fine for now.

sukesh-ak, 2022-12-05:
@hayschan As mentioned in that issue workaround, for now the fix is to add the following just before 
`esp_lcd_new_i80_bus(&bus_config, &_i80_bus);`

```c++
// add this line
bus_config.clk_src = LCD_CLK_SRC_DEFAULT;

```

file name 
16bit
`LovyanGFX\src\lgfx\v1\platforms\esp32s3\Bus_Parallel16.cpp`

8bit
`LovyanGFX\src\lgfx\v1\platforms\esp32s3\Bus_Parallel8.cpp`

hayschan, 2022-12-06:
Thanks for the help.

I tried to build again today. Before building, I make sure I download the latest code by:
1. deleting the original directory
2. `git clone --recursive https://github.com/sukesh-ak/ESP32-TUX.git`

I also changed to use ESP-IDF `v4.4.3`. But this time, the build fails. The error log is attached. It is problem related to `esp_partition`.

#### Error log
```
Building in: /Users/hays/eclipse-workspace/ESP32-TUX/build
Configuring in: /Users/hays/eclipse-workspace/ESP32-TUX/build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=/Users/hays/esp/esp-idf/tools/cmake/toolchain-esp32s3.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32s3 /Users/hays/eclipse-workspace/ESP32-TUX
-- Found Git: /usr/bin/git (found version "2.32.1 (Apple Git-133)") 
-- Component directory /Users/hays/eclipse-workspace/ESP32-TUX/components/ConfigBase does not contain a CMakeLists.txt file. No component will be added
CMake Warning at /Users/hays/esp/esp-idf/tools/cmake/project.cmake:291 (message):
  enabled ccache in build but ccache program not found
Call Stack (most recent call first):
  CMakeLists.txt:24 (project)

-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/hays/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/hays/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/hays/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- Checking Python dependencies...
Python requirements from /Users/hays/esp/esp-idf/requirements.txt are satisfied.
Processing 2 dependencies:
[1/2] espressif/fmt (9.1.0)
[2/2] idf (4.4.3)
-- LovyanGFX use components = nvs_flash;efuse;esp_lcd
CMake Error at /Users/hays/esp/esp-idf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'esp_partition'.
Call Stack (most recent call first):
  /Users/hays/esp/esp-idf/tools/cmake/build.cmake:227 (__build_resolve_and_add_req)
  /Users/hays/esp/esp-idf/tools/cmake/build.cmake:507 (__build_expand_requirements)
  /Users/hays/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:24 (project)
-- Configuring incomplete, errors occurred!
See also "/Users/hays/eclipse-workspace/ESP32-TUX/build/CMakeFiles/CMakeOutput.log".

cmake --build . -- -v
ninja: error: loading 'build.ninja': No such file or directory
Build complete (0 errors, 0 warnings): /Users/hays/eclipse-workspace/ESP32-TUX/build
Total time taken to build the project: 6,545 ms
```

This is weird. As far as I know, the `esp_partition` problem should only relate to projects that need migration to v5.0, as listed by Espressif in [their document](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/migration-guides/release-5.x/storage.html#new-component-for-the-partition-apis).

hayschan, 2022-12-06:
> If you are on `IDF 4.4.3` you can use [this version](https://github.com/sukesh-ak/ESP32-TUX/tree/569297d8924e036120ef791fa6257f6b5e2597cf) (just before IDF 5.0 upgrade)
> 
> If you are planning to do this, copy `sdkconfig.defaults.*` files from this branch which has the default values set. Or, you can make changes to the settings using `idf.py menuconfig`

Sorry, I just found out this comment. I will try this method. Thx

sukesh-ak, 2022-12-06:
> Sorry, I just found out this comment. I will try this method. Thx

Parallel related bug is fixed today by LovyanGFX. Will update the repo later today with the the fix.

hayschan, 2022-12-06:
Is it this one ? https://github.com/lovyan03/LovyanGFX/commit/e8baed4f71d6975a9aea552ef9581fc780d42420

sukesh-ak, 2022-12-06:
> Is it this one ? [lovyan03/LovyanGFX@e8baed4](https://github.com/lovyan03/LovyanGFX/commit/e8baed4f71d6975a9aea552ef9581fc780d42420)

Yes.

sukesh-ak, 2022-12-07:
@hayschan 
Project is updated with LovyanGFX fix.
Also check the readme for updated instructions including the link for Wi-Fi provisioning app.

Closing this issue since the bug is fixed and readme is updated with required instructions.

closed 2022-12-07 by sukesh-ak