sukesh-ak/ESP32-TUX #7: Error building the project

umerkk, 2022-12-13:
Hi.
I am using ESP-IDF 5, and uncommented my device in main.hpp but i am getting this error when i run idf.py menuconfig.

[image]

sukesh-ak, 2022-12-13:
Please read the steps in readme. 

You didn’t use recursive option when you cloned the repo.

umerkk, 2022-12-13:
I did.. i did pasted the command posted in the ReadME

umerkk, 2022-12-13:
[image]

[image]

sukesh-ak, 2022-12-13:
Only other reason i can think of is that your path is too long. 

Clone it in c:\github or so.

sukesh-ak, 2022-12-13:
Command for clone is this

git clone --recursive https://github.com/sukesh-ak/ESP32-TUX.git

sukesh-ak, 2022-12-13:
If its ESP32-S3 device, you need to set the target as well.

umerkk, 2022-12-13:
Yes the Long Path was an issue.. Thanks for assisting. it is working fine now.

closed 2022-12-13 by umerkk