sukesh-ak/ESP32-TUX #8: Wt32-SC01 Flash size error

umerkk, 2022-12-13:
Hi
I uncommented my device in main.hpp (WT32-SC01) which comes with a 4 MB flash.
I did changed it in the menuconfig but now i am getting the error that partition table occupies 6.6 mb and i need to change the FLASH size in menu config.

With 8MB flash size config, the project builds but the device is constantly rebooting and thats what i see in serial monitor.

`
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7000
load:0x40078000,len:15452
ho 0 tail 12 room 4
load:0x40080400,len:3840
entry 0x4008064c
I (29) boot: ESP-IDF v5.0-dirty 2nd stage bootloader
I (29) boot: compile time 12:19:56
I (29) boot: chip revision: v1.0
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 8MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 factory          factory app      00 00 00010000 00200000
I (99) boot:  4 ota_0            OTA app          00 10 00210000 00200000
I (107) boot:  5 ota_1            OTA app          00 11 00410000 00200000
I (114) boot:  6 storage          Unknown data     01 82 00610000 00080000
I (122) boot: End of partition table
I (126) boot: Defaulting to factory image
I (131) boot_comm: chip revision: 1, min. application chip revision: 0
I (138) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=a598ch (678284) map
I (392) esp_image: segment 1: paddr=000b59b4 vaddr=3ffb0000 size=04750h ( 18256) load
I (399) esp_image: segment 2: paddr=000ba10c vaddr=40080000 size=05f0ch ( 24332) load
I (409) esp_image: segment 3: paddr=000c0020 vaddr=400d0020 size=135568h (1267048) map
I (867) esp_image: segment 4: paddr=001f5590 vaddr=40085f0c size=17218h ( 94744) load
I (906) esp_image: segment 5: paddr=0020c7b0 vaddr=50000000 size=00010h (    16) load
I (921) boot: Loaded app from partition at offset 0x10000
I (921) boot: Disabling RNG early entropy source...
I (933) quad_psram: This chip is ESP32-D0WD
I (935) esp_psram: Found 8MB PSRAM device
I (935) esp_psram: Speed: 40MHz
I (937) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (945) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (954) cpu_start: Pro cpu up.
I (957) cpu_start: Starting app cpu, entry point is 0x40081778
I (950) cpu_start: App cpu up.
To exit from IDF monitor please use "Ctrl+]". Alternatively, you can use Ctrl-T Ctrl-X to exit.
To exit from IDF monitor please use "Ctrl+]". Alternatively, you can use Ctrl-T Ctrl-X to exit.
To exit from IDF monitor please use "Ctrl+]". Alternatively, you can use Ctrl-T Ctrl-X to exit.
To exit from IDF monitor please use "Ctrl+]". Alternatively, you can use Ctrl-T Ctrl-X to exit.
I (1866) esp_psram: SPI SRAM memory test OK
I (1875) cpu_start: Pro cpu start user code
I (1875) cpu_start: cpu freq: 160000000 Hz
I (1875) cpu_start: Application information:
I (1878) cpu_start: Project name:     ESP32-TUX
I (1883) cpu_start: App version:      0.16.0
I (1888) cpu_start: Compile time:     Dec 13 2022 12:18:37
I (1894) cpu_start: ELF file SHA256:  14be0ff8a7d34d0e...
ELF file not found. You need to build & flash the project before running 'monitor', and the binary on the device must match the one in the build directory exactly.
I (1900) cpu_start: ESP-IDF:          v5.0-dirty
I (1906) heap_init: Initializing. RAM available for dynamic allocation:
I (1913) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1919) heap_init: At 3FFC6B08 len 000194F8 (101 KiB): DRAM
I (1925) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1932) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1938) heap_init: At 4009D124 len 00002EDC (11 KiB): IRAM
I (1945) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (1954) spi_flash: detected chip: generic
I (1957) spi_flash: flash io: dio
E (1961) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

assert failed: do_core_init startup.c:339 (flash_ret == ESP_OK)

Backtrace: 0x4008245d:0x3ffe3a80 0x4008fe3d:0x3ffe3aa0 0x400978c1:0x3ffe3ac0 0x400d2bbb:0x3ffe3be0 0x400d2d7b:0x3ffe3c00 0x400819f2:0x3ffe3c40 0x40079611:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: 14be0ff8a7d34d0e
ELF file not found. You need to build & flash the project before running 'monitor', and the binary on the device must match the one in the build directory exactly.
`

sukesh-ak, 2022-12-14:
Project is setup with partition.csv which uses 8MB for OTA+storage. 

You might need to modify it for 4MB use. 
I will check if i can add one but probably need to remove background image to make space for OTA.

sukesh-ak, 2022-12-14:
You can set the new 4MB csv file for testing. FYI, OTA is disabled in that.
In `menuconfig` for custom parition file, set the path as `partitions/partition4MB.csv`

https://github.com/sukesh-ak/ESP32-TUX/blob/master/partitions/partition4MB.csv

closed 2022-12-14 by sukesh-ak

umerkk, 2022-12-14:
awesome.. thanks