sukesh-ak/ESP32-TUX #11: Can't run idf.py

CarbonMan, 2022-12-17:
I am following the get started instructions and have gotten to "Run idf.py menuconfig" I can't find an idf.py script.

sukesh-ak, 2022-12-17:
You need to install ESP-IDF 5.0 (Official commercial Espressif SDK)
Here is the guide for it

https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/get-started/index.html

CarbonMan, 2022-12-17:
Ah thanks
Sorry if these are ignorant questions, it looks like I have jumped in the deep end here.

I added
`
/********************DEVICE SELECTION ******************/
#define CONFIG_TUX_DEVICE_WT32_SC01
`
to main/main.hpp

Then I was able to run "idf.py build" and "idf.py -p COM4 flash" without errors, but the wt32-sc01 screen is black.

sukesh-ak, 2022-12-17:
Since you didn't mention which device I wasn't sure.

1. Grab the latest version
2. Device selection is moved to `idf.py menuconfig` now.
3. Set the device to `WT32-SC01`
4. Most probably you have 4MB version, then change partition file name to `partition-4MB.csv`
5. Change `Flash Size` to `4MB`

[image]

`idf.py menuconfig` has device selection under
```
ESP-TUX Configuration > Device Config >
```
You should be good to go with the above.

sukesh-ak, 2022-12-17:
If you have the device with 4MB Flash, the `partition-4MB` is configured without OTA support due to lack of space.

Feel free to flash a version from Web Installer to test as well. You don't need any software but just a browser.
https://tux.sukesh.me

Upgraded devices with 8MB/16MB flash are listed in the readme comparison chart. You can use that for purchase.
If you are stuck, feel free to ask.

CarbonMan, 2022-12-18:
Thank you for your help.
When doing the flash command I get
`Partitiions tables occupies 6.6MB of flash which does not fit in configured flash size 4MB.`

I then tried both the "Single factory app, no OTA" and the "Single factory app (lasrge), no OTA" options within the Partition table options. Both generate the error

`Error: app partition is too small for binary ESP32-TUX.bin size 0x1fd360`

sukesh-ak, 2022-12-18:
Check the step 4 above. There is a csv file called `partition-4MB.csv`. Use that.

CarbonMan, 2022-12-18:
Sorry,  pretty embarrasing that I completely missed that. Working beautifully now.

sukesh-ak, 2022-12-18:
> Sorry, pretty embarrasing that I completely missed that. Working beautifully now.

Glad you got it working. 
If there are issues, feel free to use `issues` here. Otherwise you can use `Discussion` option.

closed 2022-12-18 by sukesh-ak

sukesh-ak, 2023-01-07:
This is now covered in the video part of the Project Wiki here
https://github.com/sukesh-ak/ESP32-TUX/wiki