sukesh-ak/ESP32-TUX #13: Please improve the installation docs

zackees, 2022-12-29:
I'm using Windows and also mac. There are huge gabs in the readme on what to do and it seems that it was written for a veteran of ESP32 developers.

Right now I'm having trouble with the following steps:

`idf.py set-target esp32s3`
`idf.py build`
`idf.py flash monitor`

Which produce errors like the following:

```
ZacharysMBPM16:ESP32-TUX niteris$ idf.py flash monitor
Executing action: flash
Running cmake in directory /Users/niteris/dev/ESP32-TUX/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/niteris/.espressif/python_env/idf5.1_py3.10_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/niteris/dev/ESP32-TUX"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.37.1 (Apple Git-137.1)") 
-- Component directory /Users/niteris/dev/ESP32-TUX/components/ConfigBase does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/niteris/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/niteris/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/niteris/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] espressif/fmt (9.1.0)
[2/2] idf (5.1.0)
CMake Error at /Users/niteris/esp/esp-idf/tools/cmake/build.cmake:542 (message):
  ERROR:

          Some components (espressif/fmt) in the "managed_components" directory were modified         on the disk since the last run of the CMake. Content of this directory is managed automatically.

  

          If you want to keep the changes, you can move the directory with the component to the "components"         directory of your project.

  

          I.E. for "espressif__fmt" run:
          mv /Users/niteris/dev/ESP32-TUX/managed_components/espressif__fmt /Users/niteris/dev/ESP32-TUX/components/espressif__fmt

  

          Or, if you want to discard the changes remove the ".component_hash" file from the component's directory.

  

          I.E. for "espressif__fmt" run:
          rm /Users/niteris/dev/ESP32-TUX/managed_components/espressif__fmt/.component_hash
          

Call Stack (most recent call first):
  /Users/niteris/esp/esp-idf/tools/cmake/project.cmake:447 (idf_build_process)
  CMakeLists.txt:28 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/niteris/dev/ESP32-TUX/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /Users/niteris/dev/ESP32-TUX/build/log/idf_py_stderr_output_39505 and /Users/niteris/dev/ESP32-TUX/build/log/idf_py_stdout_output_39505
```

zackees, 2022-12-29:
Additionally, what is the final step necessary to flash the device once I've done all of this? Do I have to use the `idf.py` tool?

sukesh-ak, 2022-12-29:
I will work on the Wiki explaining all the steps. But for now, few tips for you.

Always install the released version of ESP-IDF which is 5.0 right now.
If you clone from the repo, you get the working version which breaks a lot and not worth the time unless you are actively contributing to Espressif development work. I believe you are running 5.1 seen from the output `[2/2] idf (5.1.0)`.

Installation instructions are here
https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32s3/get-started/index.html

This sets the controller. If you are using WT332-SC01 (which uses ESP32), then you don't need this line.
`idf.py set-target esp32s3`   

This line will build the project.   
`idf.py build`

This line will flash the firmware to the device. On Windows it detects the port by default so its not specified.   
`idf.py flash monitor`

Otherwise you can specify -p like   
`idf.py -p <your-port> flash monitor`

You can also test with the web installer from   
https://tux.sukesh.me

zackees, 2022-12-29:
Thanks this is great. I will try this.

sukesh-ak, 2022-12-29:
> Thanks this is great. I will try this.

Let me know if you have further questions. 
Will keep this issue open till you confirm you got it working.

zackees, 2022-12-29:
Thanks for your attention to this. My main device is a WT32-SCO, but while I'm traveling I'm using the WT32-SCO-Plus. Is there a change in the above instructions for the Plus?

sukesh-ak, 2022-12-29:
> idf.py set-target esp32s3

WT32-SC01 uses ESP32
WT32-SC01 Plus uses ESP32-S3

So for plus version you need to use the following target command
`idf.py set-target esp32s3`

Apart from that, make sure to select and configure the device as per README. 
`idf.py menuconfig`

Its explained here
https://github.com/sukesh-ak/ESP32-TUX#get-started

zackees, 2022-12-29:
So I followed your directions.

I installed the idf tool from the website for mac:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html

This is the setup for the v5.0.

Once installed I do a `git clone --recursive`

Then I run `idf.py set-target esp32s3` for my SCO-PLUS, I do `idf.py menuconfig` but I leave everything at the defaults.

Then when I try to do a `idf.py build` I get this:

[image]

sukesh-ak, 2022-12-29:
You at least need to select your device from `menuconfig`.  
`ESP32-TUX Configuration > Device Config`

If you still getting errors let me know and I will clone and see.

zackees, 2022-12-29:
Wow, thanks for all the quick responses. I was able to get through the last error by deleting the components.hash file (recommended by the tool itself). I'll back up through and re-do the installation and make sure Device Config is set from the menu config.

sukesh-ak, 2022-12-29:
> Wow, thanks for all the quick responses. I was able to get through the last error by deleting the components.hash file (recommended by the tool itself). I'll back up through and re-do the installation and make sure Device Config is set from the menu config.

Sounds good.

I just cloned and built it successfully.
Try to create your project inside `c:\dev` or similar with shorter folder structure.
Some of these linux tools don't like long path strings.

sukesh-ak, 2022-12-29:
Btw, `fmt` is a popular string formatter in C++ I use in IoT and Desktop projects.

https://github.com/fmtlib/fmt

zackees, 2022-12-29:
Do you have OBS? Could you do a screen record real quick of your command line?

sukesh-ak, 2022-12-29:
I do have video capture products but at the moment busy building something. Its 11:40pm here already.
If you hit errors, do let me know.

Here is what I did.
```bash

git clone --recursive https://github.com/sukesh-ak/ESP32-TUX.git
cd ESP32-TUX

idf.py set-target esp32s3

#configure/select device
idf.py menuconfig

idf.py build

```

zackees, 2022-12-29:
I'm unclear on #configure/select device that you mentioned.

What's the next step from here:

[image]

sukesh-ak, 2022-12-29:
Select your device from `menuconfig`.  
`ESP32-TUX Configuration > Device Config`

Then press `S` key to save the settings and exit. Then just `build`.

zackees, 2022-12-29:
Got it. Thanks I think this was the blocker. Will update.

zackees, 2022-12-29:
Okay the board boots up with the template!

For anyone else that's hitting this problem, this is how you select your board using the `idf.py menuconfig` step:

[image]

closed 2022-12-29 by sukesh-ak

zackees, 2022-12-29:
Do you have a consulting rate? I'd like to get a keyboard widget running so that the user can enter their wifi creds into the device. I'm willing to upstream any useful demo's back into the main template.

sukesh-ak, 2022-12-29:
> Do you have a consulting rate? I'd like to get a keyboard widget running so that the user can enter their wifi creds into the device. I'm willing to upstream any useful demo's back into the main template.

Sure. 
But Wi-Fi provisioning is already provided in the template.
I didn't add it on-device entry is because smaller screens it's difficult to type.
It's easier to do it using Smartphones. You can reset credentials and provision again easily.

Install `ESP SoftAP Provisioning` Mobile App from Espressif and scan QR code from device Settings screen to configure Wi-Fi.

[Android App](https://play.google.com/store/apps/details?id=com.espressif.provsoftap) and [iPhone App](https://apps.apple.com/in/app/esp-softap-provisioning/id1474040630)

zackees, 2022-12-30:
The goal is to not have a provisioning app on the mobile phone but instead use the on screen keyboard to enter the credentials.

What's a non public way that I communicate with you?

sukesh-ak, 2022-12-31:
> The goal is to not have a provisioning app on the mobile phone but instead use the on screen keyboard to enter the credentials.
> 
> What's a non public way that I communicate with you?

I hangout on this [FB group](https://www.facebook.com/groups/5303789986299456)
You can reach out to me on messenger.

sukesh-ak, 2023-01-10:
@zackees 
Your specific issue was related to a bug in ESP-IDF component manager. 
They are working on it I believe.

Subscribe to this issue here for updates on it
https://github.com/espressif/idf-component-manager/issues/18

zackees, 2023-01-10:
Thanks. Yeah, also it was doing this really bad because one of my computers I was using idf 5.0 and the other was 5.1 and this was causing mismatches in the hash file.