sukesh-ak/ESP32-TUX #16: WT32-SCO Broken Build - Bad GPIO define

zackees, 2023-01-03:
```bash
[1/1] cd /Users/niteris/dev/ESP32-TUX/build/bootloader/esp-idf/esptool_py && /Users/niteris/.espressif/python_env/idf5.1_py3...s/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/niteris/dev/ESP32-TUX/build/bootloader/bootloader.binBootloader binary size 0x50c0 bytes. 0x2f40 bytes (37%) free.
[5/10] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.objFAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj 
/Users/niteris/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -DESP_PLATFORM -DIDF_VER=\"v5.1-dev-2594-ga5b4e3bb3e\" -DLV_CONF_INCLUDE_SIMPLE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/niteris/dev/ESP32-TUX/build/config -I/Users/niteris/dev/ESP32-TUX/main -I/Users/niteris/dev/ESP32-TUX/main/devices -I/Users/niteris/dev/ESP32-TUX/main/helpers -I/Users/niteris/dev/ESP32-TUX/main/pages -I/Users/niteris/dev/ESP32-TUX/main/widgets -I/Users/niteris/dev/ESP32-TUX/main/images -I/Users/niteris/esp/esp-idf/components/newlib/platform_include -I/Users/niteris/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/niteris/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/niteris/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/niteris/esp/esp-idf/components/freertos/esp_additions/include -I/Users/niteris/esp/esp-idf/components/esp_hw_support/include -I/Users/niteris/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/niteris/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/Users/niteris/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/Users/niteris/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/Users/niteris/esp/esp-idf/components/heap/include -I/Users/niteris/esp/esp-idf/components/log/include -I/Users/niteris/esp/esp-idf/components/soc/include -I/Users/niteris/esp/esp-idf/components/soc/esp32s3/. -I/Users/niteris/esp/esp-idf/components/soc/esp32s3/include -I/Users/niteris/esp/esp-idf/components/hal/esp32s3/include -I/Users/niteris/esp/esp-idf/components/hal/include -I/Users/niteris/esp/esp-idf/components/hal/platform_port/include -I/Users/niteris/esp/esp-idf/components/esp_rom/include -I/Users/niteris/esp/esp-idf/components/esp_rom/include/esp32s3 -I/Users/niteris/esp/esp-idf/components/esp_rom/esp32s3 -I/Users/niteris/esp/esp-idf/components/esp_common/include -I/Users/niteris/esp/esp-idf/components/esp_system/include -I/Users/niteris/esp/esp-idf/components/esp_system/port/soc -I/Users/niteris/esp/esp-idf/components/esp_system/port/include/private -I/Users/niteris/esp/esp-idf/components/xtensa/include -I/Users/niteris/esp/esp-idf/components/xtensa/esp32s3/include -I/Users/niteris/esp/esp-idf/components/lwip/include -I/Users/niteris/esp/esp-idf/components/lwip/include/apps -I/Users/niteris/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/niteris/esp/esp-idf/components/lwip/lwip/src/include -I/Users/niteris/esp/esp-idf/components/lwip/port/esp32/include -I/Users/niteris/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/niteris/esp/esp-idf/components/json/cJSON -I/Users/niteris/dev/ESP32-TUX/components/LovyanGFX/src -I/Users/niteris/esp/esp-idf/components/nvs_flash/include -I/Users/niteris/esp/esp-idf/components/spi_flash/include -I/Users/niteris/esp/esp-idf/components/esp_partition/include -I/Users/niteris/esp/esp-idf/components/efuse/include -I/Users/niteris/esp/esp-idf/components/efuse/esp32s3/include -I/Users/niteris/esp/esp-idf/components/esp_lcd/include -I/Users/niteris/esp/esp-idf/components/esp_lcd/interface -I/Users/niteris/esp/esp-idf/components/driver/include -I/Users/niteris/esp/esp-idf/components/driver/deprecated -I/Users/niteris/esp/esp-idf/components/driver/esp32s3/include -I/Users/niteris/esp/esp-idf/components/esp_pm/include -I/Users/niteris/esp/esp-idf/components/esp_ringbuf/include -I/Users/niteris/esp/esp-idf/components/esp_timer/include -I/Users/niteris/dev/ESP32-TUX/components/lvgl -I/Users/niteris/dev/ESP32-TUX/components/lvgl/src -I/Users/niteris/dev/ESP32-TUX/components -I/Users/niteris/esp/esp-idf/components/fatfs/diskio -I/Users/niteris/esp/esp-idf/components/fatfs/vfs -I/Users/niteris/esp/esp-idf/components/fatfs/src -I/Users/niteris/esp/esp-idf/components/wear_levelling/include -I/Users/niteris/esp/esp-idf/components/sdmmc/include -I/Users/niteris/dev/ESP32-TUX/managed_components/espressif__fmt/fmt/include -I/Users/niteris/dev/ESP32-TUX/components/SettingsConfig/include -I/Users/niteris/dev/ESP32-TUX/components/OpenWeatherMap/include -I/Users/niteris/esp/esp-idf/components/app_update/include -I/Users/niteris/esp/esp-idf/components/bootloader_support/include -I/Users/niteris/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/Users/niteris/esp/esp-idf/components/esp_app_format/include -I/Users/niteris/dev/ESP32-TUX/components/ota -I/Users/niteris/esp/esp-idf/components/esp_https_ota/include -I/Users/niteris/esp/esp-idf/components/esp_http_client/include -I/Users/niteris/esp/esp-idf/components/esp_event/include -I/Users/niteris/esp/esp-idf/components/esp_wifi/include -I/Users/niteris/esp/esp-idf/components/esp_phy/include -I/Users/niteris/esp/esp-idf/components/esp_phy/esp32s3/include -I/Users/niteris/esp/esp-idf/components/esp_netif/include -I/Users/niteris/esp/esp-idf/components/wifi_provisioning/include -I/Users/niteris/esp/esp-idf/components/protocomm/include/common -I/Users/niteris/esp/esp-idf/components/protocomm/include/security -I/Users/niteris/esp/esp-idf/components/protocomm/include/transports -I/Users/niteris/esp/esp-idf/components/spiffs/include -mlongcalls  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -DLV_CONF_INCLUDE_SIMPLE=1 -I../main -Wno-missing-field-initializers -Wno-unused-variable -Wno-unused-function -Wno-pointer-arith -Og -fmacro-prefix-map=/Users/niteris/dev/ESP32-TUX=. -fmacro-prefix-map=/Users/niteris/esp/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -fdiagnostics-color=always -std=gnu++20 -fno-exceptions -fno-rtti -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj -c /Users/niteris/dev/ESP32-TUX/main/main.cpp
In file included from /Users/niteris/dev/ESP32-TUX/main/main.hpp:62,
                 from /Users/niteris/dev/ESP32-TUX/main/main.cpp:26:
/Users/niteris/dev/ESP32-TUX/main/devices/conf_WT32SCO1.h: In constructor 'LGFX::LGFX()':
/Users/niteris/dev/ESP32-TUX/main/devices/conf_WT32SCO1.h:49:21: error: 'GPIO_NUM_22' was not declared in this scope; did you mean 'GPIO_NUM_42'?
   49 | #define TFT_RST     GPIO_NUM_22
      |                     ^~~~~~~~~~~
/Users/niteris/dev/ESP32-TUX/main/devices/conf_WT32SCO1.h:49:21: note: in definition of macro 'TFT_RST'
   49 | #define TFT_RST     GPIO_NUM_22
      |                     ^~~~~~~~~~~
In file included from /Users/niteris/dev/ESP32-TUX/components/lvgl/src/core/lv_obj.h:140,
                 from /Users/niteris/dev/ESP32-TUX/components/lvgl/lvgl.h:35,
                 from /Users/niteris/dev/ESP32-TUX/main/helpers/helper_display.hpp:27,
                 from /Users/niteris/dev/ESP32-TUX/main/main.hpp:78,
                 from /Users/niteris/dev/ESP32-TUX/main/main.cpp:26:
/Users/niteris/dev/ESP32-TUX/components/lvgl/src/core/lv_obj_style.h: In function 'void lv_obj_remove_style_all(_lv_obj_t*)':
/Users/niteris/dev/ESP32-TUX/components/lvgl/src/core/lv_obj_style.h:94:48: warning: bitwise operation between different enumeration types '<unnamed enum>' and '<unnamed enum>' is deprecated [-Wdeprecated-enum-enum-conversion]
   94 |     lv_obj_remove_style(obj, NULL, LV_PART_ANY | LV_STATE_ANY);
      |                                    ~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from /Users/niteris/dev/ESP32-TUX/main/main.hpp:92,
                 from /Users/niteris/dev/ESP32-TUX/main/main.cpp:26:
/Users/niteris/dev/ESP32-TUX/main/gui.hpp: In function 'void create_footer(lv_obj_t*)':
/Users/niteris/dev/ESP32-TUX/main/gui.hpp:401:63: warning: bitwise operation between different enumeration types '<unnamed enum>' and '<unnamed enum>' is deprecated [-Wdeprecated-enum-enum-conversion]
  401 |     lv_obj_add_style(footerButtons, &style_glow,LV_PART_ITEMS | LV_BTNMATRIX_CTRL_CHECKED); // selected
      |                                                 ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/niteris/dev/ESP32-TUX/build/log/idf_py_stderr_output_42757 and /Users/niteris/dev/ESP32-TUX/build/log/idf_py_stdout_output_42757
```

sukesh-ak, 2023-01-03:
Please select the correct device from menuconfig.

sukesh-ak, 2023-01-03:
According to the device selected in menuconfig, the device configuration header file changes.
Relevant code which does that is here

https://github.com/sukesh-ak/ESP32-TUX/blob/c4efb5980bd16e067c8496db6997336b9035f49c/main/main.hpp#L59

sukesh-ak, 2023-01-05:
Please refer Wiki for more details on how to configure your device to use ESP32-TUX
https://github.com/sukesh-ak/ESP32-TUX/wiki

closed 2023-01-05 by sukesh-ak