sukesh-ak/ESP32-TUX #18: Web installer fail

eried, 2023-01-21:
Hi! I am trying to test this using the web installer on sc01 plus but I get:
`Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL.`

I only have a reset button, is there is something simple I am failing to do?

sukesh-ak, 2023-01-21:
This is an issue related to ESP32-S3 with native USB support.
You can join here on our fb group where people have provided workarounds for this issue.
https://www.facebook.com/groups/5303789986299456

Alternatively, I would suggest to get the debugger for WT32-SC01 Plus from the link in the readme of this repo.
This will ease the pain.

sukesh-ak, 2023-01-21:
@eried Here is the short answer
[image]

eried, 2023-01-21:
Oh brilliant, thanks a lot!

closed 2023-01-21 by eried