sukesh-ak/ESP32-TUX #19: template arduino framework

mess-maker, 2023-01-24:
Hi,
do you plan to make template like this one with arduino framework ?

Graphics & Touch Driver : [LovyanGFX](https://github.com/lovyan03/LovyanGFX)
UI / Widgets : [LVGL 8.x](https://github.com/lvgl/lvgl)
Framework : arduino

just received my new WT32-SC01 Plus, and don't know how to get started for now as I prefer to stick to arduino framework for now

sukesh-ak, 2023-01-25:
@mess-maker 
This project was a base template I created for one of my commercial requirements. 
Due to Arduino license restrictions, I used `ESP-IDF` for it.

`ESP-IDF` in C++ is same as `Arduino`.

```c++
/* ESP-IDF code for Arduino */

// Your arduino code
void setup() {

}

void loop() {

}

// This portion is merged when you compile Arduino, that's all.
// This is part of one of the files in the Arduino installed folder
extern "C" void app_main(void) {

	setup();  // Your Arduino setup function
	while (1) {
		loop();  // Your Arduino loop function
	}
}

```

```

closed 2023-01-25 by sukesh-ak

mess-maker, 2023-01-28:
thanks I'll try to get started with that

knopserl, 2023-06-06:
> 
this is a great project for the WT32-SC01 Plus, but I also wanted to use it in the Arduino IDE. 
It it feasible and worth to spent the effort to migrate this ESP-IDF  based code to an Arduino IDE acceppted code?
I have tried to insert the main.cpp code in the "voide setup() function and an empto loop function (as I have not seen a matching or required loop code) and it does of course not compile because it does not even accept the include files such as #include <spi_flash_mmap.h> ?
Any help possible or to much to change?