sukesh-ak/ESP32-TUX #24: WT32-SC01 - Only shows a white screen

ericjmonson, 2023-03-18:
I've gotten the demo to work great for a WT32-SC01+ board. However, when I follow the following video step by step, I end up with just a white screen on the display after flashing:
https://asciinema.org/a/549415

I have set the board up in "idf.py menuconfig" AS A "Wireless-Tag WT32-SC01" ... see the screen scrape from idf.py menuconfig:

--------------------
(Top) → ESP32-TUX Configuration → Device Config
                                                                                             Espressif IoT Development Framework Configuration
    <choice TUX_DEVICE_SELECTED> Select the device (Wireless-Tag WT32-SC01 - SPI - 4MB Flash)  --->
(partitions/partition-4MB.csv) <PARTITION_TABLE_CUSTOM_FILENAME> Custom Partition CSV filename
    <choice ESPTOOLPY_FLASHSIZE> Flash size (4 MB)  --->
[*] <WALLPAPER_IMAGE> Enable wallpaper image
--------------------

sukesh-ak, 2023-03-19:
@ericjmonson 
Can you try flashing your device from the following URL for testing?
https://tux.sukesh.me

Also, if you have compiled for Plus version, please make sure to delete the build folder and start fresh since the target device settings might be wrong.
`WT32-SC01` uses ESP32
`WT32-SC01 Plus` uses ESP32-S3

I would suggest, you use a released version of ESP-IDF since the main branch of the repo is a moving target and is expected to have breaking changes.

ericjmonson, 2023-03-20:
That web installer is pretty slick. Unfortunately, though, I still only see a white screen after flashing with your version. I happen to have a handful of the WT32-SC01 (original ... not plus) devices, so I tried it on two different models. Both end up with just a white screen after flashing.

As for building, I was worried about artifacts lying around. I did a completely fresh clone of the git repo, made sure the esp32 was set for the build target, followed the video for setting up the partitions and selecting the device. I end up with a white screen when I compile on my own.

sukesh-ak, 2023-03-20:
> That web installer is pretty slick. Unfortunately, though, I still only see a white screen after flashing with your version. I happen to have a handful of the WT32-SC01 (original ... not plus) devices, so I tried it on two different models. Both end up with just a white screen after flashing.
> 
> As for building, I was worried about artifacts lying around. I did a completely fresh clone of the git repo, made sure that esp32 was set as the build target, followed the video for setting up the partitions and selecting the device. I end up with a white screen when I compile on my own.

I will test and revert.

gjm19761, 2023-03-23:
i have exactly same issue wt32-sc01 and get white screen after flashing and reboot, and using web installer

sukesh-ak, 2023-03-23:
> i have exactly same issue wt32-sc01 and get white screen after flashing and reboot, and using web installer

Will check and update tomorrow. Sorry for the delay.

sukesh-ak, 2023-03-24:
Update: @ericjmonson & @gjm19761 

I just flashed my WT32-SC01 device with firmware from https://tux.sukesh.me
And it works perfectly fine as expected.

So possible reasons for you not working
- Wrong device?
- Device specs have changed (like display controller)
- Not genuine device?

Can't think of any other reason :(

If you haven't checked yet, please flash from this website using 1st option and check.
https://tux.sukesh.me

gjm19761, 2023-03-24:
same issue using  first option on website, my device isnt a new one (i bought it last year) , mine is a wt32-sc01-v3.2

sukesh-ak, 2023-03-24:
> same issue using first option on website, my device isnt a new one (i bought it last year) , mine is a wt32-sc01-v3.2

You can check with this older sample without LVGL `using the code in the README`.
https://github.com/sukesh-ak/LVGL8-WT32-SC01-Arduino

ericjmonson, 2023-03-24:
Sorry for the sideways video ... but so you can see it happening in action:
https://www.youtube.com/watch?v=91B7wnW03ww

I definately picked the correct one. I also have a wt32-sc01-v3.2 (as you'll see from the video).

I compiled and flashed the arduino version above (https://github.com/sukesh-ak/LVGL8-WT32-SC01-Arduino#touch-and-draw-sample-without-lvgl). When it starts up, the screen flashes white then black a number of times, then eventually just turns black.

sukesh-ak, 2023-03-24:
> Sorry for the sideways video ... but so you can see it happening in action: https://www.youtube.com/watch?v=91B7wnW03ww
>
> I definately picked the correct one. I also have a wt32-sc01-v3.2 (as you'll [...]

Can you check this from Arduino?
Please install `LovyanGFX` library and select `ESP32 Dev` Module for devices.

https://gist.github.com/sukesh-ak/a00d4c7d41a469d341972b3727e5029e

ericjmonson, 2023-03-24:
Yeah ... as I mentioned above, I did install LovyanGFS in the Ardunio IDE and selected ESP32 Dev as the board. That's where I get the black and white flashes followed by a black screen.

sukesh-ak, 2023-03-24:
> Yeah ... as I mentioned above, I did install LovyanGFS in the Ardunio IDE and selected ESP32 Dev as the board. That's where I get the black and white flashes followed by a black screen.

ok try the above snippet I just created to narrow down the issue.

ericjmonson, 2023-03-24:
Flashing the above application, I end up with a white screen.

https://gist.github.com/sukesh-ak/a00d4c7d41a469d341972b3727e5029e

ericjmonson, 2023-03-25:
Well ... good news. Been playing around with this today. Was able to get it working with my version of the boards. Had to change one line:
main/devices/conf_WT32SCO1.h
Line 81: cfg.freq_write = 80000000; // SPI clock on transmission (up to 80MHz, rounded to 80MHz divided by integer)
Changed to: cfg.freq_write = 40000000;

Now its all working. Not sure if there is more noise on the line making the SPI bus unstable at that clock speed, or why I had to drop the SPI clock speed.

gjm19761, 2023-03-25:
cool  where is that file? i suspect wireless tag released a different version of the sc01 so now theres another variantto deal with

gjm19761, 2023-03-25:
> Well ... good news. Been playing around with this today. Was able to get it working with my version of the boards. Had to change one line: main/devices/conf_WT32SCO1.h Line 81: cfg.freq_write = [...]

Can confirm this also fixed mine, will this be added to the web installer as suspect theres been a revision of the sc01

sukesh-ak, 2023-03-29:
Glad you guys got it sorted it out :)

closed 2023-04-12 by sukesh-ak

jrversteegh, 2024-11-13:
@sukesh-ak This issue was closed, but not fixed?

sukesh-ak, 2024-11-13:
> @sukesh-ak This issue was closed, but not fixed?

Haven't got time to confirm if that's the fix since it was working fine on my device.
Currently too busy to get all the devices checked again since ESP-IDF versions are being updated as well.

jrversteegh, 2024-11-15:
OK, fair enough. Thanks for a great template!

But maybe leave the issue open (and potentially ask for a pull request to fix it if you don't have time yourself) rather than close the issue without a fix? I was at least confused, because I expected it to be fixed when it wasn't..
In either case, I can confirm the issue on wt32-sc01-v3.2. What version is "your device"?