sukesh-ak/ESP32-TUX #25: Failed build

igormoochnick, 2023-03-24:
Most likely I'm missing a step but, after doing:

```
bringing all submodules in
idf.py set-target esp32
idf.py build
```
I get the following error:
```
/Dev/CNC/ESP32-TUX/main/main.hpp:73:6: error: #error Unsupported device. Configure device in menuconfig
   73 |     #error Unsupported device. Configure device in menuconfig
      |      ^~~~~
In file included from /Dev/CNC/ESP32-TUX/main/main.hpp:51:
 /Dev/CNC/ESP32-TUX/main/helpers/helper_sntp.hpp: In function 'void obtain_time()':
/Dev/CNC/ESP32-TUX/main/helpers/helper_sntp.hpp:112:25: warning: 'void sntp_servermode_dhcp(int)' is deprecated: use esp_sntp_servermode_dhcp() instead [-Wdeprecated-declarations]
  112 |     sntp_servermode_dhcp(1);
      |     ~~~~~~~~~~~~~~~~~~~~^~~
In file included from /Dev/CNC/ESP32-TUX/main/helpers/helper_sntp.hpp:27:
/Dev/CNC/esp-idf/components/lwip/include/apps/esp_sntp.h:241:6: note: declared here
  241 | void sntp_servermode_dhcp(int set_servers_from_dhcp)
      |      ^~~~~~~~~~~~~~~~~~~~
/Dev/CNC/ESP32-TUX/main/helpers/helper_sntp.hpp: In function 'void initialize_sntp()':
/Dev/CNC/ESP32-TUX/main/helpers/helper_sntp.hpp:133:26: warning: 'void sntp_setoperatingmode(u8_t)' is deprecated: use esp_sntp_setoperatingmode() instead [-Wdeprecated-declarations]
  133 |     sntp_setoperatingmode(SNTP_OPMODE_POLL);
      |     ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
/Dev/CNC/esp-idf/components/lwip/include/apps/esp_sntp.h:235:6: note: declared here
  235 | void sntp_setoperatingmode(u8_t operating_mode)
      |      ^~~~~~~~~~~~~~~~~~~~~
/Dev/CNC/ESP32-TUX/main/helpers/helper_sntp.hpp:134:23: warning: 'void sntp_setservername(u8_t, const char*)' is deprecated: use esp_sntp_setservername() instead [-Wdeprecated-declarations]
  134 |     sntp_setservername(0, "pool.ntp.org");
      |     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
/Dev/CNC/esp-idf/components/lwip/include/apps/esp_sntp.h:249:6: note: declared here
  249 | void sntp_setservername(u8_t idx, const char *server)
      |      ^~~~~~~~~~~~~~~~~~
/Dev/CNC/ESP32-TUX/main/helpers/helper_sntp.hpp:139:14: warning: 'void sntp_init()' is deprecated: use esp_sntp_init() instead [-Wdeprecated-declarations]
  139 |     sntp_init();
      |     ~~~~~~~~~^~
/Dev/CNC/esp-idf/components/lwip/include/apps/esp_sntp.h:255:6: note: declared here
  255 | void sntp_init(void)
      |      ^~~~~~~~~
In file included from /Dev/CNC/ESP32-TUX/components/lvgl/src/core/lv_obj.h:140,
                 from /Dev/CNC/ESP32-TUX/components/lvgl/lvgl.h:35,
                 from /Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp:27,
                 from /Dev/CNC/ESP32-TUX/main/main.hpp:78:
/Dev/CNC/ESP32-TUX/components/lvgl/src/core/lv_obj_style.h: In function 'void lv_obj_remove_style_all(_lv_obj_t*)':
/Dev/CNC/ESP32-TUX/components/lvgl/src/core/lv_obj_style.h:94:48: warning: bitwise operation between different enumeration types '<unnamed enum>' and '<unnamed enum>' is deprecated [-Wdeprecated-enum-enum-conversion]
   94 |     lv_obj_remove_style(obj, NULL, LV_PART_ANY | LV_STATE_ANY);
      |                                    ~~~~~~~~~~~~^~~~~~~~~~~~~~
/Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp: At global scope:
/Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp:58:37: error: 'TFT_WIDTH' was not declared in this scope
   58 | static const uint16_t screenWidth = TFT_WIDTH;
      |                                     ^~~~~~~~~
/Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp:59:38: error: 'TFT_HEIGHT' was not declared in this scope; did you mean 'FOOTER_HEIGHT'?
   59 | static const uint16_t screenHeight = TFT_HEIGHT;
      |                                      ^~~~~~~~~~
      |                                      FOOTER_HEIGHT
/Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp:70:8: error: 'LGFX' does not name a type
   70 | static LGFX lcd; // declare display variable
      |        ^~~~
/Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp: In function 'esp_err_t lv_display_init()':
/Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp:91:5: error: 'lcd' was not declared in this scope
   91 |     lcd.setRotation(2);
      |     ^~~
/Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp:129:11: error: 'esp_timer_create_args_t' does not name a type
  129 |     const esp_timer_create_args_t lv_periodic_timer_args = {
      |           ^~~~~~~~~~~~~~~~~~~~~~~
/Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp:136:5: error: 'esp_timer_handle_t' was not declared in this scope; did you mean 'lv_timer_handler'?
  136 |     esp_timer_handle_t lv_periodic_timer;
      |     ^~~~~~~~~~~~~~~~~~
      |     lv_timer_handler
[... 56 lines cut ...]
/Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp: In function 'void touchpad_read(lv_indev_drv_t*, lv_indev_data_t*)':
/Dev/CNC/ESP32-TUX/main/helpers/helper_display.hpp:237:20: error: 'lcd' was not declared in this scope
  237 |     bool touched = lcd.getTouch(&touchX, &touchY);
      |                    ^~~
In file included from /Dev/CNC/ESP32-TUX/main/main.hpp:92:
/Dev/CNC/ESP32-TUX/main/gui.hpp: In function 'void create_footer(lv_obj_t*)':
/Dev/CNC/ESP32-TUX/main/gui.hpp:401:63: warning: bitwise operation between different enumeration types '<unnamed enum>' and '<unnamed enum>' is deprecated [-Wdeprecated-enum-enum-conversion]
  401 |     lv_obj_add_style(footerButtons, &style_glow,LV_PART_ITEMS | LV_BTNMATRIX_CTRL_CHECKED); // selected
      |                                                 ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
/Dev/CNC/ESP32-TUX/main/gui.hpp: In function 'void slider_event_cb(lv_event_t*)':
/Dev/CNC/ESP32-TUX/main/gui.hpp:495:5: error: 'lcd' was not declared in this scope
  495 |     lcd.setBrightness((int)lv_slider_get_value(slider));
      |     ^~~
/Dev/CNC/ESP32-TUX/main/gui.hpp: In function 'void tux_panel_config(lv_obj_t*)':
/Dev/CNC/ESP32-TUX/main/gui.hpp:515:60: error: 'lcd' was not declared in this scope
  515 |     lv_label_set_text_fmt(slider_label, "Brightness : %d", lcd.getBrightness());
      |                                                            ^~~
/Dev/CNC/ESP32-TUX/main/main.cpp: In function 'void app_main()':
/Dev/CNC/ESP32-TUX/main/main.cpp:269:5: error: 'lcd' was not declared in this scope
  269 |     lcd.init();         // Initialize LovyanGFX
      |     ^~~
[1085/1165] Building CXX object esp-idf...yanGFX.dir/src/lgfx/v1/LGFXBase.cpp.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Dev/CNC/ESP32-TUX/build/log/idf_py_stderr_output_47987 and /Dev/CNC/ESP32-TUX/build/log/idf_py_stdout_output_47987
```
Any suggestions?

sukesh-ak, 2023-03-24:
> 

Did you check the video walkthrough on the steps required available in the readme?

igormoochnick, 2023-03-24:
Yes. Followed video exactly

igormoochnick, 2023-03-24:
I have a feeling that the Lovyan package has changed and the dependency brings a wrong version

sukesh-ak, 2023-03-24:
> I have a feeling that the Lovyan package has changed and the dependency brings a wrong version

I will check during the weekend 🤞🏼

EmbeddedTec, 2023-04-02:
Wich version if ESP-IDF are you using?
I ran into the same problem using ESP-IDF v5.1-dev-3619-g57b6be22a7

It compiles without errors using the release version ESP-IDF v5.0.1

sukesh-ak, 2023-04-12:
Always use release branch since main is moving target which has build issues sometimes.

closed 2023-04-12 by sukesh-ak