sukesh-ak/ESP32-TUX #29: Include Path errors

fnandomaia, 2023-05-17:
Hello.
I´m using Visual Studio Code. 
The code is building fine, but I see errors as shown bellow.
What´s the directory I need to set in INCLUDE PATH for  C++ Extension, or is there any configuration to fix these errors? 

[image]
[image]

sukesh-ak, 2023-05-17:
I think your ESP-IDF vscode extension is not configured properly. 

You can goto extension settings and rerun the configuration.

fnandomaia, 2023-05-17:
Fixed including full path of includes. 

Before:
#include "OpenWeatherMap.hpp"
After:
#include "../components/OpenWeatherMap/include/OpenWeatherMap.hpp"

closed 2023-05-17 by fnandomaia