sukesh-ak/ESP32-TUX #32: Opportunities for shared modules/models

cydergoth, 2023-06-27:
https://github.com/cydergoth/tembed-app-shell

Similar project for the T-Embed S3

cydergoth, 2023-06-28:
> Are you asking for help or showcasing something you made?

Suggesting we can share a common core of widget and BT code etc. Both apps
are similar enough it should be possible to factor out commonalities

sukesh-ak, 2023-08-01:
> Suggesting we can share a common core of widget and BT code etc. Both apps are similar enough it should be possible to factor out commonalities
> […](#)

Sorry been busy with work.

I wanted to design a UI different from ESP32-TUX for smaller screens.
I have not got time to work on it yet but the repo was created here
https://github.com/TUX-Templates/ESP32-TUX-Mini

closed 2023-08-01 by sukesh-ak