sukesh-ak/ESP32-TUX #33: Build fails

veljko, 2023-07-23:
Hi and thanks for this great project!
This is related to #25 which has been closed, but as 5.1 is release and I get the same error with 5.0.3 when running idf.py build 😕

C:/code/iot/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/misc/pixelcopy.cpp: In constructor 'lgfx::v1::pixelcopy_t::pixelcopy_t(const void*, lgfx::v1::color_depth_t, lgfx::v1::color_depth_t, bool, const void*, uint32_t)':
C:/code/iot/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/misc/pixelcopy.cpp:39:25: error: member 'lgfx::v1::pixelcopy_t::<anonymous>' is used uninitialized [-Werror=uninitialized]
   39 |     , src_mask  ( (1 << src_bits) - 1 )
      |                         ^~~~~~~~
C:/code/iot/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/misc/pixelcopy.cpp:40:25: error: member 'lgfx::v1::pixelcopy_t::<anonymous>' is used uninitialized [-Werror=uninitialized]
   40 |     , dst_mask  ( (1 << dst_bits) - 1 )

Thanks!

sukesh-ak, 2023-07-30:
Looks like an issue with LovyanGFX. Please update that submodule in the project and see if it fixed your problem. 

I am a bit busy with other projects so cannot promise when I can test it myself.

veljko, 2023-08-03:
Thanks Sukesh. After bumping the LovyanGFX and lvgl to the latest release, everything is fine.

closed 2023-08-03 by veljko