sukesh-ak/ESP32-TUX #39: using the wrong file .......... picks up esp32 when esp32s3 is required

SoniaZotz, 2023-10-17:
[image]

sukesh-ak, 2023-10-19:
Check the video in the readme for how to change the MCU for ESP-IDF.
https://github.com/sukesh-ak/ESP32-TUX#watch-project-demo-walkthrough---clone-build-and-flash

If you are new to ESP-IDF, I would suggest to learn a bit more, than creating new issues on this repo.

closed 2023-10-19 by sukesh-ak

SoniaZotz, 2023-10-19:
I suggest  that you try to compile this git repo using esp-idf-v5.1.1\
please run it for me and see the results 

as for the little video - it would be simpler just to have a text list of steps 

I appreciate  your work, experience and knowledge and please be gentle we all started to walk with someone's help 
Thank you

git clone --recursive https://github.com/sukesh-ak/ESP32-TUX.git
cd E:\__test\ESP32-TUX
idf.py set-target esp32s3

-- --------------Compile Info------------
-- IDF_PATH = E:/Espressif/frameworks/esp-idf-v5.1.1
-- IDF_TARGET = esp32s3
-- PROJECT_NAME = ESP32-TUX
-- PROJECT_DIR = E:/__test/ESP32-TUX
-- BUILD_DIR = E:/__test/ESP32-TUX/build
-- SDKCONFIG = E:/__test/ESP32-TUX/sdkconfig
-- SDKCONFIG_DEFAULTS = E:/__test/ESP32-TUX/sdkconfig.defaults
-- CONFIG_LV_CONF_SKIP =
-- ---------------------------------------
-- CMAKE_SOURCE_DIR = E:/__test/ESP32-TUX
-- CMAKE_BINARY_DIR = E:/__test/ESP32-TUX/build
-- ---------By Sukesh Ashok Kumar---------
-- Configuring done
-- Generating done
-- Build files have been written to: E:/__test/ESP32-TUX/build

=======================================================================================================
idf.py menuconfig

 Select the device (MakerFabs ESP32S3SPI35 - SPI - 16MB Flash)  --->
   Flash size (16 MB)  --->  
 Configuration saved to 'E:/__test/ESP32-TUX/sdkconfig'  
======================================================================================================= 
idf.py build

-- --------------Compile Info------------
-- IDF_PATH = E:/Espressif/frameworks/esp-idf-v5.1.1
-- IDF_TARGET = esp32s3
-- PROJECT_NAME = ESP32-TUX
-- PROJECT_DIR = E:/__test/ESP32-TUX
-- BUILD_DIR = E:/__test/ESP32-TUX/build
-- SDKCONFIG = E:/__test/ESP32-TUX/sdkconfig
-- SDKCONFIG_DEFAULTS = E:/__test/ESP32-TUX/sdkconfig.defaults
-- CONFIG_LV_CONF_SKIP =
-- ---------------------------------------
-- CMAKE_SOURCE_DIR = E:/__test/ESP32-TUX
-- CMAKE_BINARY_DIR = E:/__test/ESP32-TUX/build
-- ---------By Sukesh Ashok Kumar---------
-- Configuring done
-- Generating done
-- Build files have been written to: E:/__test/ESP32-TUX/build
[5/1213] Generating ../../partition_table/partition-table.binPartition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,2M,
ota_0,app,ota_0,0x210000,2M,
ota_1,app,ota_1,0x410000,2M,
storage,data,spiffs,0x610000,512K,

=======================================================================================================
E:/__test/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/platforms/esp32/common.cpp:812:14: error: 'struct i2c_dev_t' has no member named 'comd0'; did you mean 'comd'?
  812 |       (&dev->comd0)[index].val = cmd_val;
      |              ^~~~~
[... 38 lines cut ...]
E:/Espressif/frameworks/esp-idf-v5.1.1/components/lwip/include/apps/esp_sntp.h:249:6: note: declared here
  249 | void sntp_setservername(u8_t idx, const char *server)
      |      ^~~~~~~~~~~~~~~~~~
E:/__test/ESP32-TUX/main/helpers/helper_sntp.hpp:139:14: warning: 'void sntp_init()' is deprecated: use esp_sntp_init() instead [-Wdeprecated-declarations]
  139 |     sntp_init();
      |     ~~~~~~~~~^~
E:/Espressif/frameworks/esp-idf-v5.1.1/components/lwip/include/apps/esp_sntp.h:255:6: note: declared here
  255 | void sntp_init(void)
      |      ^~~~~~~~~
In file included from E:/__test/ESP32-TUX/components/lvgl/src/core/lv_obj.h:140,
                 from E:/__test/ESP32-TUX/components/lvgl/lvgl.h:35,
                 from E:/__test/ESP32-TUX/main/helpers/helper_display.hpp:27,
                 from E:/__test/ESP32-TUX/main/main.hpp:78:
E:/__test/ESP32-TUX/components/lvgl/src/core/lv_obj_style.h: In function 'void lv_obj_remove_style_all(_lv_obj_t*)':
E:/__test/ESP32-TUX/components/lvgl/src/core/lv_obj_style.h:94:48: warning: bitwise operation between different enumeration types '<unnamed enum>' and '<unnamed enum>' is deprecated [-Wdeprecated-enum-enum-conversion]
   94 |     lv_obj_remove_style(obj, NULL, LV_PART_ANY | LV_STATE_ANY);
      |                                    ~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from E:/__test/ESP32-TUX/main/main.hpp:92:
E:/__test/ESP32-TUX/main/gui.hpp: In function 'void create_footer(lv_obj_t*)':
E:/__test/ESP32-TUX/main/gui.hpp:401:63: warning: bitwise operation between different enumeration types '<unnamed enum>' and '<unnamed enum>' is deprecated [-Wdeprecated-enum-enum-conversion]
  401 |     lv_obj_add_style(footerButtons, &style_glow,LV_PART_ITEMS | LV_BTNMATRIX_CTRL_CHECKED); // selected
      |                                                 ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the E:\__test\ESP32-TUX\build\log\idf_py_stderr_output_25064 and E:\__test\ESP32-TUX\build\log\idf_py_stdout_output_25064
PS E:\__test\ESP32-TUX>