sukesh-ak/ESP32-TUX #40: Please confirm the compile

SoniaZotz, 2023-10-19:
I suggest that you try to compile this git repo using esp-idf-v5.1.1
please run it for me and see the results

as for the little video - it would be simpler just to have a text list of steps

I appreciate your work, experience and knowledge and please be gentle we all started to walk with someone's help
Thank you

git clone --recursive https://github.com/sukesh-ak/ESP32-TUX.git
cd E:__test\ESP32-TUX
idf.py set-target esp32s3

-- --------------Compile Info------------
-- IDF_PATH = E:/Espressif/frameworks/esp-idf-v5.1.1
-- IDF_TARGET = esp32s3
-- PROJECT_NAME = ESP32-TUX
-- PROJECT_DIR = E:/__test/ESP32-TUX
-- BUILD_DIR = E:/__test/ESP32-TUX/build
-- SDKCONFIG = E:/__test/ESP32-TUX/sdkconfig
-- SDKCONFIG_DEFAULTS = E:/__test/ESP32-TUX/sdkconfig.defaults
-- CONFIG_LV_CONF_SKIP =

-- CMAKE_SOURCE_DIR = E:/__test/ESP32-TUX
-- CMAKE_BINARY_DIR = E:/__test/ESP32-TUX/build
-- ---------By Sukesh Ashok Kumar---------
-- Configuring done
-- Generating done
-- Build files have been written to: E:/__test/ESP32-TUX/build

=======================================================================================================
idf.py menuconfig
Select the device (MakerFabs ESP32S3SPI35 - SPI - 16MB Flash) --->
Flash size (16 MB) --->
Configuration saved to 'E:/__test/ESP32-TUX/sdkconfig'

idf.py build

-- --------------Compile Info------------
-- IDF_PATH = E:/Espressif/frameworks/esp-idf-v5.1.1
-- IDF_TARGET = esp32s3
-- PROJECT_NAME = ESP32-TUX
-- PROJECT_DIR = E:/__test/ESP32-TUX
-- BUILD_DIR = E:/__test/ESP32-TUX/build
-- SDKCONFIG = E:/__test/ESP32-TUX/sdkconfig
-- SDKCONFIG_DEFAULTS = E:/__test/ESP32-TUX/sdkconfig.defaults
-- CONFIG_LV_CONF_SKIP =

-- CMAKE_SOURCE_DIR = E:/__test/ESP32-TUX
-- CMAKE_BINARY_DIR = E:/__test/ESP32-TUX/build
-- ---------By Sukesh Ashok Kumar---------
-- Configuring done
-- Generating done
-- Build files have been written to: E:/__test/ESP32-TUX/build
[5/1213] Generating ../../partition_table/partition-table.binPartition table binary generated. Contents:
ESP-IDF Partition Table
Name, Type, SubType, Offset, Size, Flags

nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,2M,
ota_0,app,ota_0,0x210000,2M,
ota_1,app,ota_1,0x410000,2M,
storage,data,spiffs,0x610000,512K,

=======================================================================================================
E:/__test/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/platforms/esp32/common.cpp:812:14: error: 'struct i2c_dev_t' has no member named 'comd0'; did you mean 'comd'?
812 | (&dev->comd0)[index].val = cmd_val;
| ^~~~~
| comd
[1094/1213] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/extra/libs/fsdrv/lv_fs_posix.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the E:__test\

=============================================================================================================
E:/__test/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/misc/pixelcopy.cpp:39:25: error: member 'lgfx::v1::pixelcopy_t::' is used uninitialized [-Werror=uninitialized]
39 | , src_mask ( (1 << src_bits) - 1 )
| ^~~~~~~~
E:/__test/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/misc/pixelcopy.cpp:40:25: error: member 'lgfx::v1::pixelcopy_t::' is used uninitialized [-Werror=uninitialized]
40 | , dst_mask ( (1 << dst_bits) - 1 )
| ^~~~~~~~
cc1plus.exe: some warnings being treated as errors
[109/110] Generating binary image from built executableesptool.py v4.7.dev2
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Generated E:/__test/ESP32-TUX/build/bootloader/bootloader.bin
[110/110] cmd.exe /C "cd /D E:__test\ESP32-TUX\build\boot...r 0x0 E:/__test/ESP32-TUX/build/bootloader/bootloader.bin"Bootloader binary size 0x5200 bytes. 0x2e00 bytes (36%) free.
[37/255] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.objIn file included from E:/__test/ESP32-TUX/main/main.hpp:51,
from E:/__test/ESP32-TUX/main/main.cpp:26:
E:/__test/ESP32-TUX/main/helpers/helper_sntp.hpp: In function 'void obtain_time()':
E:/__test/ESP32-TUX/main/helpers/helper_sntp.hpp:112:25: warning: 'void sntp_servermode_dhcp(int)' is deprecated: use esp_sntp_servermode_dhcp() instead [-Wdeprecated-declarations]
112 | sntp_servermode_dhcp(1);
| ~~~~~~~~~~~~~~~~~~~~^~~
In file included from E:/__test/ESP32-TUX/main/helpers/helper_sntp.hpp:27:
E:/Espressif/frameworks/esp-idf-v5.1.1/components/lwip/include/apps/esp_sntp.h:241:6: note: declared here
241 | void sntp_servermode_dhcp(int set_servers_from_dhcp)
| ^~~~~~~~~~~~~~~~~~~~
E:/__test/ESP32-TUX/main/helpers/helper_sntp.hpp: In function 'void initialize_sntp()':
E:/__test/ESP32-TUX/main/helpers/helper_sntp.hpp:133:26: warning: 'void sntp_setoperatingmode(u8_t)' is deprecated: use esp_sntp_setoperatingmode() instead [-Wdeprecated-declarations]
133 | sntp_setoperatingmode(SNTP_OPMODE_POLL);
| ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
E:/Espressif/frameworks/esp-idf-v5.1.1/components/lwip/include/apps/esp_sntp.h:235:6: note: declared here
235 | void sntp_setoperatingmode(u8_t operating_mode)
| ^~~~~~~~~~~~~~~~~~~~~
E:/__test/ESP32-TUX/main/helpers/helper_sntp.hpp:134:23: warning: 'void sntp_setservername(u8_t, const char*)' is deprecated: use esp_sntp_setservername() instead [-Wdeprecated-declarations]
134 | sntp_setservername(0, "pool.ntp.org");
| ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
E:/Espressif/frameworks/esp-idf-v5.1.1/components/lwip/include/apps/esp_sntp.h:249:6: note: declared here
249 | void sntp_setservername(u8_t idx, const char server)
| ^~~~~~~~~~~~~~~~~~
E:/__test/ESP32-TUX/main/helpers/helper_sntp.hpp:139:14: warning: 'void sntp_init()' is deprecated: use esp_sntp_init() instead [-Wdeprecated-declarations]
139 | sntp_init();
| ~~~~~~~~~^~
E:/Espressif/frameworks/esp-idf-v5.1.1/components/lwip/include/apps/esp_sntp.h:255:6: note: declared here
255 | void sntp_init(void)
| ^~~~~~~~~
In file included from E:/__test/ESP32-TUX/components/lvgl/src/core/lv_obj.h:140,
from E:/__test/ESP32-TUX/components/lvgl/lvgl.h:35,
from E:/__test/ESP32-TUX/main/helpers/helper_display.hpp:27,
from E:/__test/ESP32-TUX/main/main.hpp:78:
E:/__test/ESP32-TUX/components/lvgl/src/core/lv_obj_style.h: In function 'void lv_obj_remove_style_all(_lv_obj_t)':
E:/__test/ESP32-TUX/components/lvgl/src/core/lv_obj_style.h:94:48: warning: bitwise operation between different enumeration types '' and '' is deprecated [-Wdeprecated-enum-enum-conversion]
94 | lv_obj_remove_style(obj, NULL, LV_PART_ANY | LV_STATE_ANY);
| ~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from E:/__test/ESP32-TUX/main/main.hpp:92:
E:/__test/ESP32-TUX/main/gui.hpp: In function 'void create_footer(lv_obj_t*)':
E:/__test/ESP32-TUX/main/gui.hpp:401:63: warning: bitwise operation between different enumeration types '' and '' is deprecated [-Wdeprecated-enum-enum-conversion]
401 | lv_obj_add_style(footerButtons, &style_glow,LV_PART_ITEMS | LV_BTNMATRIX_CTRL_CHECKED); // selected
| ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the E:__test\ESP32-TUX\build\log\idf_py_stderr_output_25064 and E:__test\ESP32-TUX\build\log\idf_py_stdout_output_25064
PS E:__test\ESP32-TUX>

sukesh-ak, 2023-10-21:
It was built with ESP-IDF 5. If you have newer SDK version, there might be breaking changes in the SDK.

Currently I don't have time to upgrade IDF and test. 
From your output I see `deprecated` messages which means IDF has breaking changes which needs to be fixed.

SoniaZotz, 2023-10-21:
https://github.com/SoniaZotz/ESP32-TUX

closed 2023-12-30 by sukesh-ak