sukesh-ak/ESP32-TUX #43: Hardware display rotation

THE-KONDRAT, 2023-10-27:
Hi. I want to set default orientation as landscape.
I tried to change conf_WT32SCO1.h (width and height in defines) and helper_display.hpp (lcd.setRotation(3 instead of 2) ). Orientation changes but display width crops by 320 pixels.
I using WT32-SC01.
How can i change display orientation?

sukesh-ak, 2023-10-27:
> Hi. I want to set default orientation as landscape. I tried to change conf_WT32SCO1.h (width and height in defines) and helper_display.hpp (lcd.setRotation(3 instead of 2) ). Orientation changes but display width crops by 320 pixels. I using WT32-SC01. How can i change display orientation?

Display default orientation is setup here. Try changing and see if it works for your requirements
https://github.com/sukesh-ak/ESP32-TUX/blob/37822a7b6dfcb67e278c78b2aa468817b6fa0bbe/main/helpers/helper_display.hpp#L91

THE-KONDRAT, 2023-10-27:
With `lcd.setRotation(2)` display orientation is portrait.
[image]

With `lcd.setRotation(1)` display orientation is landscape but with black area on the right.
[image]

If i rotate display (with `lcd.setRotation(2)`) from ui it works.
How can i fix that?

sukesh-ak, 2023-10-27:
> With `lcd.setRotation(2)` display orientation is portrait. [image]
> 
> With `lcd.setRotation(1)` display orientation is landscape but with black area on the right. [image]
> 
> If i rotate display (with `lcd.setRotation(2)`) from ui it works. How can i fix that?

Background image needs to be changed to cover the side. 
Guess you figured that out yourself.

THE-KONDRAT, 2023-10-27:
Ok. I temporary disabled background image, but the black area still exists. Looks like the image width is 320 pix instead of 480.

sukesh-ak, 2023-12-30:
> It works. I changed the width and height in helper and set rotation 1 instead of 3. And i set sw_rotate=0. But it doesn't look right. Is there another way to rotate the picture?

You might need 2 images, one for each orientation and dynamically load it.

closed 2023-12-30 by sukesh-ak