sukesh-ak/ESP32-TUX #49: WaveShare ESP32-S3-LCD-4.3

janick, 2024-01-04:
Would it be possible to have the main/devices/... file for the new WaveShare ESP32-S3-LCD-4.3??

https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3

janick, 2024-01-04:
Hmmmm..... Don't think it will be easy: the display uses the ST7262 driver and there is not LGFX Panel_LCD implementation for it.

sukesh-ak, 2024-01-04:
Its an RGB interface I guess. You can open a request on LovyanGFX repo for support of this.

sukesh-ak, 2024-01-04:
Have added a support request for this on LovyanGFX
https://github.com/lovyan03/LovyanGFX/issues/502

janick, 2024-01-04:
Looks like someone already has an implementation: https://github.com/lovyan03/LovyanGFX/issues/240

closed 2024-01-04 by sukesh-ak

janick, 2024-01-04:
> Samples are here for both Arduino & ESP-IDF https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3#demo1

But they are TFT_eSPI based, not LogyanGFX.

sukesh-ak, 2024-01-04:
Using RGB panel configuration, it should work with LovyanGFX

Westcott1, 2024-03-02:
I have this device working with Arduino_GFX.
However my attempt at creating an RGB panel results in the ESP32 crashing at gfx.init();
Also. I don't know what to set these to -
      cfg.pin_henable = GPIO_NUM_41;
      cfg.pin_vsync = GPIO_NUM_40;
      cfg.pin_hsync = GPIO_NUM_39;

```
#define LGFX_USE_V1
#include <LovyanGFX.hpp>
#include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>
#include <driver/i2c.h>

class LGFX : public lgfx::LGFX_Device {
public:
  lgfx::Bus_RGB _bus_instance;
  lgfx::Panel_RGB _panel_instance;
  lgfx::Light_PWM _light_instance;
  lgfx::Touch_GT911 _touch_instance;
  LGFX(void) {
    {
      auto cfg = _panel_instance.config();
      cfg.memory_width = TFT_HOR_RES;
      cfg.memory_height = TFT_VER_RES;
      cfg.panel_width = TFT_HOR_RES;
      cfg.panel_height = TFT_VER_RES;
      cfg.offset_x = 0;
      cfg.offset_y = 0;
      _panel_instance.config(cfg);
    }

    {
      auto cfg = _bus_instance.config();
      cfg.panel = &_panel_instance;

      cfg.pin_d0 = 14;  // B0
      cfg.pin_d1 = 38;   // B1
      cfg.pin_d2 = 18;   // B2
      cfg.pin_d3 = 17;   // B3
      cfg.pin_d4 = 10;   // B4

      cfg.pin_d5 = 37;   // G0
      cfg.pin_d6 = 0;  // G1
      cfg.pin_d7 = 45;   // G2
      cfg.pin_d8 = 48;   // G3
      cfg.pin_d9 = 47;  // G4
      cfg.pin_d10 = 21;  // G5

      cfg.pin_d11 = 1;  // R0
      cfg.pin_d12 = 2;  // R1
      cfg.pin_d13 = 42;  // R2
      cfg.pin_d14 = 41;  // R3
      cfg.pin_d15 = 40;  // R4

      cfg.pin_henable = GPIO_NUM_41;
      cfg.pin_vsync = GPIO_NUM_40;
      cfg.pin_hsync = GPIO_NUM_39;
      cfg.pin_pclk = 1;
      cfg.freq_write = 16000000;

      cfg.hsync_polarity = 0;
      cfg.hsync_front_porch = 40;
      cfg.hsync_pulse_width = 48;
      cfg.hsync_back_porch = 88;

      cfg.vsync_polarity = 0;
      cfg.vsync_front_porch = 11;
      cfg.vsync_pulse_width = 3;
      cfg.vsync_back_porch = 32;

      cfg.pclk_active_neg = 1;
      cfg.de_idle_high = 0;
      cfg.pclk_idle_high = 0;

      _bus_instance.config(cfg);
    }
    _panel_instance.setBus(&_bus_instance);

    {
      auto cfg = _light_instance.config();
      cfg.pin_bl = GPIO_NUM_2;
      _light_instance.config(cfg);
    }
    _panel_instance.light(&_light_instance);

    {
      auto cfg = _touch_instance.config();
      cfg.x_min = 0;
      cfg.x_max = TFT_HOR_RES-1;
      cfg.y_min = 0;
      cfg.y_max = TFT_VER_RES-1;
      cfg.pin_int = 4;
      cfg.pin_rst = -1;
      cfg.bus_shared = false;
      cfg.offset_rotation = 0;
      cfg.i2c_port = I2C_NUM_1;
      cfg.pin_sda = 8;
      cfg.pin_scl = 9;
      cfg.freq = 400000;
      cfg.i2c_addr = 0x14;
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);
    }
    setPanel(&_panel_instance);
  }
};

```

sukesh-ak, 2024-03-03:
@Westcott1 
I would suggest you to ask on `LovyanGFX` repo for your config.