sukesh-ak/ESP32-TUX #51: build errors

joekam54, 2024-02-21:
Hi!
First Thanks for the great work you are doing! this is a verynice project.

I managed to compile for WT32-SC01 3.5" and flashed it
Ineede to change my computer in the meantime and got all the setup new the same way, but Iam getting now compilation errors.
I fresh cloned your project, I installed esp-idf-version 5.2 -> the same I have used before.

I also tried it with idf Version 5.0 but the same result. no successfull build
[idf_py_stdout_output_36856.log](https://github.com/sukesh-ak/ESP32-TUX/files/14362137/idf_py_stdout_output_36856.log)
 
I have the build-log attached, pls help
Thanks
Regards
Joe

sukesh-ak, 2024-02-21:
You are using `IDF 5.2` and `LovyanGFX` has some bug which is fixed for this version.
You can read about it [here](https://github.com/lovyan03/LovyanGFX/issues/489)

You can try using `IDF 5.1.2` which should work fine for now.
More details and fix for `IDF 5.2` is in the above link.

closed 2024-02-21 by sukesh-ak

joekam54, 2024-02-22:
Hi Sukesh,

Thank you for the fast attention and reply!

 

I have tried with idf 5.1.2 but iam getting another build error.

 

/Users/Admin/Documents/A-TUX-ESP/ESP32-TUX/main/main.cpp: In function 'void app_main()':

C:/Users/Admin/Documents/A-TUX-ESP/ESP32-TUX/main/main.cpp:269:5: error: 'lcd' was not declared in this scope

  269 |     lcd.init();         // Initialize LovyanGFX

      |     ^~~

ninja: build stopped: subcommand failed.

Log attached.

 

Kind Regards

Joe

sukesh-ak, 2024-02-22:
> C:/Users/Admin/Documents/A-TUX-ESP/
Try not to use long folder tree since some of the compiler tools create issues.

Looks like `LovyanGFX` library access is missing (best guess).  
Maybe you didn't get the submodules while cloning.

joekam54, 2024-02-23:
Hi Sukesh,

Thanks , it works now! 😊

 

Kind regards

Joe

joekam54, 2024-02-23:
> @joekam54 <https://github.com/joekam54>
> If you want a simple example just for your TFT WT32-SC01 you can try this new sample.
> I am trying to get lvgl v9 working for this sample. But the current version is using v8.
>
> https://github.com/sukesh-ak/IDF-ESP_LCD-LVGL

…nice.. thanks for help getting me into esp-idf..

 

I will set up idf an VS-code on my linux PC, because I really hate to work on windows 

KR